airo-ugent/airo-mono #52: Awaitable from `move_linear_to_tcp_pose` does not finish.

Victorlouisdg, 2023-04-19:
Awaitable from `move_linear_to_tcp_pose` does not finish, it always waits 10 seconds, then times out.

I've commited a [quick fix](https://github.com/airo-ugent/airo-mono/commit/a1f4d40d9e16c2fad3daae77f71172a590c9b6f9) to main to remove the `_is_move_command_finished` check.

The source of the problem is that `getAsyncOperationProgress` keeps returning 0, even when the TCP pose is really close to the requested TCP pose. This might be a problem on the control box itself, so outside of our control.

The problem occurs for both robots. It does not occur after a `moveJ`  command. Following up a `moveL` with a `moveJ` without checking the `getAsyncOperationProgress` seems to work as expected.

Victorlouisdg, 2024-01-02:
We should probably also undo the quick fix from the first comment and test this.

tlpss, 2024-01-12:
fixed in 59ca0a7bf12c59388b3ffec3b868c5167a307956

closed 2024-01-12 by tlpss