airo-ugent/airo-mono #56: [airo-camera-toolkit] record tcp poses of calibration to autorun with planner

adverley, 2023-04-24:
When the planner is fully functional and reliable, we could autocalibrate predefined tcp poses for calibrating cameras.

tlpss, 2023-04-24:
I think that only works for an approx known camera pose (the marker has to be in view) and environment (some poses might be collisions for the tcp or the board). But we could definitely teleop/guide the robot a first time and then use those poses with the planner for future calibrations in the same setup?

adverley, 2023-04-24:
Fully agreed. It helps to replicate existing setups (e.g. suppose the
camera on setup A for experiment X is remounted for some reasons).  Other
example; we bumped into the side camera a couple of times for unfoldir on
victorlouise.

Victorlouisdg, 2023-12-11:
Running the calibration now stores the TCP poses to disk as `calibration_<datetime>/data/tcp_pose_0000.json`, docs [here](https://github.com/airo-ugent/airo-mono/tree/main/airo-camera-toolkit/airo_camera_toolkit/calibration).

tlpss, 2024-01-02:
> Running the calibration now stores the TCP poses to disk as `calibration_<datetime>/data/tcp_pose_0000.json`, docs [here](https://github.com/airo-ugent/airo-mono/tree/main/airo-camera-toolkit/airo_camera_toolkit/calibration).

closing this

closed 2024-01-02 by tlpss