airo-ugent/airo-mono #78: Port simple OpenCV annotation tool

Victorlouisdg, 2023-08-01:
For the ICRA23 competition I wrote [a simple tool](https://github.com/Victorlouisdg/linen/blob/main/linen/linen/opencv/annotation.py) to collect a few simple annotations from manual clicks: 
[image]

The idea is that you only need to provide a camera and an annotation specification and do a single function call:
```python
annotation_spec = {
    "keypoint": Annotation.Keypoint,
    "line_segment": Annotation.LineSegment,
    "line": Annotation.Line,
    "bounding_box": Annotation.BoundingBox,
}
annotations = get_manual_annotations(camera, annotation_spec)
```
One change that should be made is instead of using a `RGBCamera` object, the `get_manual_annotations()` function should simply take an image.

We also need to decide where this script goes. Perhaps in `airo-dataset-tools` or `airo-camera-toolkit`?

tlpss, 2023-08-02:
Maybe to `camera-toolkit`? I think we will mostly use this to bypass perception modules. But both are fine by me as long as it is documented

Victorlouisdg, 2023-08-29:
Implemented in PR #82.

closed 2023-08-29 by Victorlouisdg