airo-ugent/airo-mono #103: move from python 3.8-3.10 to 3.10-12

tlpss, 2024-01-02:
3.8 will be EOL in oktober, So I would argue we can move to >=3.10. which we can enforce in the python setup.py

would also need to change tests here: https://github.com/airo-ugent/airo-mono/blob/5cbacd7acee7491435f963e39b33ea29af6b4357/.github/workflows/pytest.yaml#L17

tlpss, 2024-01-03:
@Victorlouisdg  @adverley  anyone has an issue with this proposed change?

tlpss, 2024-01-03:
also need to change the conda environment's default python version

Victorlouisdg, 2024-01-03:
Sounds good to me, however some packages like [ur_rtde](https://gitlab.com/sdurobotics/ur_rtde) only have prebuilt wheels up to 3.11 I believe, making it difficult to install on 3.12.  

We should probably also update our types too then, because 3.10 allow `tuple` instead of `Tuple` etc.

PR #112 (2024-01-12): update supported python versions
supported python versions from 3.8-3.10 to 3.9-3.11

closes #103
closed 2024-01-12 by tlpss