airo-ugent/airo-mono #116: Realsense cannot be instantiated more than once with multiprocessing

m-decoster, 2024-01-26:
**Describe the bug**

`RuntimeError: Frame didn't arrive within 5000`

This is the same issue as #109 which was fixed in #115 for single process code.

The fix does not work when used with the `multiprocessing` package

**To Reproduce**

Below script:

```
# Set up multiprocessing camera.
namespace = "camera"

# Creating and starting the publisher
p = MultiprocessRGBPublisher(
    Realsense,
    camera_kwargs={
        "resolution": Realsense.RESOLUTION_720,
        "fps": 30,
        "enable_hole_filling": True,
    },
    shared_memory_namespace=namespace,
)
p.start()

self._camera = MultiprocessRGBReceiver(namespace)
self._rerun_logger = MultiprocessRGBRerunLogger(namespace)
self._rerun_logger.start()
```

Realsense will fail to send frames

**Expected behavior**

This should start and log images to rerun, but it only does so after replugging the USB cable (as was the case in the original issue)

tlpss, 2024-02-02:
hm weird that it no longer works for the multiprocessing case. Could it be due to faulty resource management somehow? (i.e. closing the receivers without properly closing the camera?) @Victorlouisdg  any thoughts?

tlpss, 2024-02-02:
maybe related to #98

Victorlouisdg, 2024-02-03:
There's a few things we can investigate:
* are you sure the publisher process has terminated and thus released the handle on the camera hardware?
* if it was terminated: was the `__exit__` method of the `RealSense` object called and thus `pipeline.stop()`?
* if that was called, is the devices listed in the call to `devices = ctx.query_devices()` used by the fix?

m-decoster, 2024-12-06:
I haven't encountered this issue any more since the `spawn` multiprocessing method is used.

closed 2024-12-06 by m-decoster