airo-ugent/airo-mono #121: ur_rtde's prebuilt wheels are not compatible with Python 3.11

m-decoster, 2024-02-01:
**Describe the bug**

When trying to use airo-robots from a conda environment with Python 3.11, the import of `rtde_control` fails:

```
File "~/.conda/envs/airo-mono/lib/python3.11/site-packages/airo_robots/manipulators/hardware/ur_rtde.py", line 10, in <module>
    from rtde_control import RTDEControlInterface
ModuleNotFoundError: No module named 'rtde_control'
```
This is a known issue: https://gitlab.com/sdurobotics/ur_rtde/-/issues/294

Apparently, building the wheel from source fixes this, but this requires a CMake installation.

I'm raising this issue because airo-mono's environment.yaml file specifies Python 3.11 (though it works fine on 3.10). I discovered it because I was trying to use some functionality that only became available in Python 3.11.

**To reproduce**

1. Create and activate a new conda environment with Python 3.11 and pip as dependencies
2. Run `pip install ur_rtde`
3. Within the python REPL, run `import rtde_control`

**Possible solutions**

Use Python 3.10, build `ur_rtde` from source, or wait for [this issue](https://gitlab.com/sdurobotics/ur_rtde/-/issues/294) to be fixed to use Python 3.11.

tlpss, 2024-02-02:
@m-decoster  nice catch, maybe best if we revert the python version to 3.10 for now and wait for the binaries to be pushed to PyPI? Feel free to make that change if you agree :)

m-decoster, 2024-05-23:
I can confirm that the [underlying issue](https://gitlab.com/sdurobotics/ur_rtde/-/issues/294) has been fixed. Nothing needs to be done on our end, but it's now possible again to use Python 3.11 with ur_rtde.