airo-ugent/airo-mono #125: `MultiprocessRGB(D)Publisher` cannot insatiated `Zed2i` camera

Victorlouisdg, 2024-02-09:
**Describe the bug**
The publisher process hangs when initializing a `Zed2i` camera.

**To Reproduce**
Run one of the multiprocessing scripts. 

**Expected behavior**
Yesterday multiprocessing still worked on my setup, I even recorded some short videos with the `MultiprocessVideoRecorder`.

**Environment:**
Gorilla desktop. Tested with two Zed2i camera. Occasionally got the error code `NO GPU DETECTED`. Rebooted desktop multiple times.

Victorlouisdg, 2024-02-09:
Creating a `Zed2i` camera in a regular Python script / parent process still works fine. So this has something to do with the fact that a subprocess started from a Python script behaves differently than its parent process.

m-decoster, 2024-02-09:
May be related to #116

Victorlouisdg, 2024-02-12:
The probleem seem to be GPU availablity in the Publisher process. Trying to close the camera results in:
```
CUDA error at Camera.cpp:163 code=304(cudaErrorOperatingSystem) "void sl::Camera::close()" 
```
I also get:
```
IndexError: Could not open Zed2i camera, error = NO GPU DETECTED
```
PyTorch also does not find any GPUs in the created process `torch.cuda.is_available()` returns `False`.

Victorlouisdg, 2024-02-12:
Adding `multiprocessing.set_start_method("spawn")` fixes the CUDA issue and allows opening ZED cameras. However, the ZED SDK always starts optimizing the neural depth mode with this start method.

Victorlouisdg, 2024-02-12:
Nevermind, I was confused about the neural_depth optimization. It started because I changed my CUDA version to 12.3, and is not related to multiprocessing.

Victorlouisdg, 2024-02-12:
This fixes the issue, but requires you to do this in every script with a `MultiprocessPublisher`:
```python  
multiprocessing.set_start_method("spawn") 
```
A more elegant solution is inheriting from: 
```python
multiprocessing.context.SpawnProcess
```
But that does not work and I don't understand why.

Victorlouisdg, 2024-02-19:
Another downside to `"spawn"` vs the default `"fork"` method, is that it can be much slower, as it copies the entire memory of the parent process (can be several seconds vs several ms). (However in this case this copy is probably what prevents the CUDA errors.)

m-decoster, 2024-12-06:
> A more elegant solution is inheriting from:
> 
> ```python
> multiprocessing.context.SpawnProcess
> ```
> 
> But that does not work and I don't understand why.

My experience is that for many libraries, including the Zed SDK, PyTorch, and open3d, you indeed need to use the `spawn` method, probably because they all depend on CUDA (e.g., using open3d's CPU functionality *does* work with `fork`). But, for me simply inheriting from `multiprocessing.context.SpawnProcess` does work, even when not using `multiprocessing.set_start_method("spawn")`.

I'll be doing a lot of multiprocessing stuff in the near future, I'll report here if I find any issues when not using `multiprocessing.set_start_method("spawn") `.