airo-ugent/airo-mono #134: Point cloud to world frame transformation

m-decoster, 2024-02-29:
**Describe the feature you'd like**
When working with point clouds, sometimes we want to express them in the world (or robot base) frame.
I currently do this with a method proposed by @Victorlouisdg through Open3D:

```
pcd_in_camera = point_cloud_to_open3d(point_cloud)
pcd = pcd_in_camera.transform(X_W_C)
point_cloud = open3d_to_point_cloud(pcd)
```

Do we want to create a function for this in the camera toolkit (leaving possibilities for future optimisations, I haven't checked the performance implications yet of the above conversions to and from Open3D. I'm assuming it's fairly cheap because it's zero copy)?

**Use cases**
- Merging point clouds from different cameras (e.g., with ICP)
- Logging point clouds to, e.g., rerun

**Possible implementation**
We could wrap the above code snippet into a function named `transform_point_cloud`, which could go into https://github.com/airo-ugent/airo-mono/blob/main/airo-camera-toolkit/airo_camera_toolkit/point_clouds/operations.py

tlpss, 2024-02-29:
Actually you can do the following w/o going to open3D:

```
# assumes pc is an Nx4 array with homogeneous points. 
pc_in_A = (X_A_B @ pc_in_B.T).T
```

In terms of adding a function for this, we could do that if required/desired.

I think the pc operations are indeed the best place to add them.  There is actually already a helper class in the `spatial-algebra` package, that could be used, we can simply wrap it: https://github.com/airo-ugent/airo-mono/blob/main/airo-spatial-algebra/airo_spatial_algebra/operations.py.

m-decoster, 2024-02-29:
So if I get it correctly, we would only need to wrap `operations.transform_points`, because the point cloud positions is a `Vectors3DType`.

Since Open3D conversion is not required in that case, then perhaps adding an additional function for this is overkill.

(For me this was about reducing a 3-line operation + imports to a one-liner, but `transform_points` does the same)

tlpss, 2024-02-29:
> So if I get it correctly, we would only need to wrap operations.transform_points, because the point cloud positions is a Vectors3DType.

indeed!

> Since Open3D conversion is not required in that case, then perhaps adding an additional function for this is overkill.

on the other hand it will be easier to find. So I think it could still be a good idea

PR #141 (2024-03-07): Point cloud transform
## Describe your changes

fix #134 

## Checklist
- [x] Have you modified the changelog?
- [ ] If you made changes to the hardware interfaces, have you tested them using the manual tests?
closed 2024-03-07 by m-decoster