airo-ugent/airo-mono #142: `BoundingBox3DType` is not generic over frames in which the bounding box is used

m-decoster, 2024-03-15:
In some cases, we want to define a bounding box in the object's local frame, as this makes it easy to reason about `minimal` and `maximal` points that define the `BoundingBox3DType`.

It is of course possible to do this, but if we then want to use this bounding box in another frame, e.g., to log it in rerun or to add it to a simulation in drake, we need to transform the bounding box points.

We cannot simply transform the two points with [`transform_points`](https://github.com/airo-ugent/airo-mono/blob/main/airo-spatial-algebra/airo_spatial_algebra/operations.py), because minimum and maximum points in one frame may not be the minimum and maximum points in another frame.

One solution is to compute the eight corners of the bounding box, transform these, and recompute the minimum and maximum points. For example:

```python
bounding_box_in_frame_A: BoundingBox3DType = get_bounding_box_in_frame_A()
corners_in_frame_A = compute_corners_of_bounding_box(bounding_box_in_frame_A)
corners_in_frame_B = transform_points(X_B_A, corners_in_frame_A)
bounding_box_in_frame_B = np.min(corners_in_frame_B, axis=0), np.max(corners_in_frame_B, axis=0)
```

We could create a function that performs these operations. Alternatively, is there some representation of a 3D bounding box that does not have this issue but is still easy to work with?

m-decoster, 2024-03-15:
I am a proponent of creating a function `transform_bounding_box` analogous to `transform_point_cloud` (#141), unless someone thinks another definition for `BoundingBox3DType` is in order.

tlpss, 2024-03-19:
I agree that this is a useful 'tool' to have. 
I'm not an expert so I'll leave the details to you :) 

However, I think it is important to 
1. make sure there can be no confusion about which type of bbox is being used, so the typing/documentation should reflect this.
2. leverage existing libraries as much as possible!  W/o good reason, I think we should try to use open3D as much as possible. I can agree with some custom types for bounding boxes if they are being used in multiple places across airo-mono, but other than that we should leverage open3D as much as possible imo.

m-decoster, 2024-03-19:
That is a fair point @tlpss. In airo-mono itself, `BoundingBox3dType` is currently only used to crop point clouds. I also use it downstream for barista, and I see it's also used in several places in the cloth competition.

I want to raise this as a discussion point, either with @Victorlouisdg directly, or in our upcoming meeting. I'll add it to the agenda in any case.

m-decoster, 2024-08-13:
> leverage existing libraries as much as possible! W/o good reason, I think we should try to use open3D as much as possible. I can agree with some custom types for bounding boxes if they are being [...]

5 months later, I see no other use cases in airo-mono for OBBs, in applications I've been mostly using Open3D directly. My proposal is to close this issue and keep `BoundingBox3DType` for AABBs only, until and if the need arises to support OBBs in airo-mono.

closed 2024-08-13 by m-decoster