airo-ugent/airo-mono #143: Catch RTDE exceptions in URrtde constructor

Victorlouisdg, 2024-03-28:
Sometimes connecting the to robot fails with weird errors. This often happens after protective stops or emergency stops. Restarting your script usually solves this. However it would be nice if airo-mono tries a few times itself, e.g:
```python
max_attempts = 3
for i in range(max_attempts):
    try:
        gripper_left = Robotiq2F85(ip_address_left)
        robot_left = URrtde(ip_address_left, URrtde.UR3E_CONFIG, gripper_left)

        gripper_right = Robotiq2F85(ip_address_right)
        robot_right = URrtde(ip_address_right, URrtde.UR3E_CONFIG, gripper_right)

        dual_arm = DualArmPositionManipulator(robot_left, X_W_LCB, robot_right, X_W_RCB)
        return dual_arm
    except RuntimeError as e:
        print(f"Failed to connect to UR5e robots. Attempt {i+1}/{max_attempts}. Error:\n {e}")
        if i == max_attempts - 1:
            raise e
        time.sleep(2)
```
This works for me for the cloth competition, example of it saving me a little time:
[image]

tlpss, 2024-03-28:
I think we should only add this to the 'real' hardware classes? not to the Bimanual interfaces on top of them?

Victorlouisdg, 2024-03-28:
Agreed.

PR #148 (2024-10-25): Closes #143 - Catch RTDE exceptions in constructor
## Describe your changes

When the RTDE connection fails to establish, `URrtde` will now retry up to three times.
This is useful after emergency stops, when the connection sometimes fails.

Closes #143 

## Checklist
- [x] Have you modified the changelog?
- [x] If you made changes to the hardware interfaces, have you tested them using the manual tests?
commit 6fc4a677 (2024-10-25): Closes #143 - Catch RTDE exceptions in constructor (#148)

* Retry RTDE connection multiple times upon failure
    * Break from loop on successful connection
    * Sleep on failure, to give connection attempt some time to reset
closed 2024-10-25 by m-decoster