airo-ugent/airo-mono #149: [airo-typing] InverseKinematicsFunctionType could support current joint configuration as input for diff IK

adverley, 2024-11-29:
**Describe the feature you'd like**
[InverseKinematicsFunctionType](https://github.com/airo-ugent/airo-mono/blob/6fc4a677a34e535b62c87eb07847e10b2015f78f/airo-typing/airo_typing/__init__.py#L123) could support an optional current joint configuration as input for differential IK. Or do we want an DifferentialInverseKinematicsFunctionType? This would feel like the start of many more variations which we want to avoid. 

If we add JointConfigurationType to the Callable type, does this break existing implementations where IK functions only support one input argument? 

Anyways, airo-planner uses this type in the [planner definition](https://github.com/airo-ugent/airo-planner/blob/1f2ebeb181afad6700e41d1a97173e2ea7f4bcf4/airo_planner/ompl/single_arm_ompl_planner.py#L47) which prevents me from using diff IK which is linearized around the current joint config. 

**Use cases**
Anyone using diff IK instead of IK. 

**Possible implementation**
`InverseKinematicsFunctionType = Callable[[HomogeneousMatrixType, Optional[JointConfigurationType]], List[JointConfigurationType]]`

or new member:
`DifferentialInverseKinematicsFunctionType = Callable[[HomogeneousMatrixType, Optional[JointConfigurationType]], List[JointConfigurationType]]`

tlpss, 2024-12-02:
I'm not sure when you want to use diff IK in our current motion planning, but I'm not the expert :stuck_out_tongue:   I think we always plan in joint config space at the moment? 

I think you can update the function type to include the optional joint config, should not break things but it might be safe to run some hardware tests before merging? 

pinging @Victorlouisdg, since he wrote most of the motion planning features.

Victorlouisdg, 2024-12-02:
The philosophy used in `airo-planner` is to use _functions with signatures that are as simple as possible as interface_. 
 This means it accepts functions that take only a TCP pose as input. This keeps the internals of `airo-planner` simple.  The same "simple signature" principle applies to `is_state_valid_fn` and mimics how OMPL works.

If you have a more complex IK function signature, you can use this code pattern to simplify it before passing it to `airo-planner`:
```python
from functools import partial 

def fancy_ik_fn(tcp_pose, closest_joints, extra_param):
    # ...IK logic using all arguments ...
    return joint_angles 

simple_ik_fn = partial(fancy_ik_fn, closest_joints=q, extra_param=0.5) 

planner = SingleArmOmplPlanner(collision_checker_fn, inverse_kinematics_fn=simple_ik_fn) 
```

Since `SingleArmOmplPlanner` is stateless, you can modify its `inverse_kinematics_fn` attribute at any time. This allows you to adapt the planner's behaviour without creating a new instance:

```python
planner = SingleArmOmplPlanner(...)

new_ik_fn = partial(fancy_ik_fn, closest_joints=q_current)

planner.inverse_kinematics_fn = new_ik_fn 

planner.plan_to_tcp_pose(...)
```

This design also provides flexibility if we want to integrate with other IK libraries that may require additional parameters in the future.

tlpss, 2024-12-02:
sidenote: a different (more object-oriented) approach would be to create a class for the IK solver and have method that runs the IK and only takes a target pose, you can then do whatever updates you want/  store a state /... in the class instance.

tlpss, 2024-12-02:
@Victorlouisdg , your opinion is to not update the Callable interface to include the optional current configuration, as it would not be used by OMPL by default?

And so if you want to run more complex IK, you have to perform all logic outside of OMPL, options include:
- create a class for the IK implementation, 
- use Partial() to wrap any fixed arguments 
- or change the callable object manually in between calling the planner?

adverley, 2024-12-02:
Thanks for the suggestions and code samples, they are very useful!

There are two main reasons for using the current joint configuration in IK: (1) we often want an IK solution that is close to the current joint configuration, (2) if you use diff IK, you need the current joint configuration anyway. I understand that for (1) you can return a list of IK solutions and filter yourself (as you do in the analytical IK) but not all IK solvers work that way. 

I really like the KISS principle in airo-mono (and airo-planner) so that is why I am raising this thread. For me, resorting to partials actually increases the complexity for my application because the underlying airo-mono wants to keep it simple. But I do want to avoid opening the flood gates for future function arguments (joint limits, solve time, etc) although I think many of those arguments can be stateful in the IK solver itself as it does not change when calling `inverse_kinematics(...)` (e.g. joint limits do not change every call). Any thoughts about this?

tlpss, 2024-12-02:
I think we might want to first list the use cases we want to support and then discuss which additions we want? 

Bc as usual, I'll be on the fences about overscoping the airo-mono functionalities, as that is not what is was intended for :wink: 

We can do this in-person and provide a brief summary here to backtrack our decisions in the future!

m-decoster, 2025-01-13:
Conclusions from Friday: we want to avoid creating a large multitude of types for IK, and instead are in favour of creating a class and implementing `__call__`, if it is feasible to maintain state this way.

(Correct me if I'm wrong, my memory on this is a bit fuzzy)