airo-ugent/airo-mono #158: Upgrade Numpy to version >= 2.0

m-decoster, 2025-05-26:
**Describe the feature you'd like**
As more packages downstream are upgrading to Numpy 2.0, we should also consider upgrading.

**Use cases**
I've noticed that some libraries that we depend on have started porting to Numpy 2.0, and this can lead to conflicts in the package manager that need to be resolved manually.

**Possible implementation**
We should test in an environment with a recent Numpy 2.0 version to see if everything still works, and if necessary, update code to conform with the new API. Things to check:

- Do all manual tests still work?
- Does upgrading to Numpy >= 2.0 conflict with any of our dependencies?

m-decoster, 2025-07-03:
Possibly requires rewriting the calibration routines, since OpenCV versions compatible with NumPy 2.0 deprecated some of the functions we use: e.g. https://docs.opencv.org/4.11.0/d9/d6a/group__aruco.html#ga21b51b9e8c6422a4bac27e48fa0a150b

tlpss, 2025-07-03:
Agreed that we need to migrate!

The opencv thing is annoying because some of those require 'HW integration tests'. 

I think this is mainly:
- arm, gripper and camera implementations
- camera calibration methods

we can coordinate in this PR which one of those have been tested?